geerlingguy/3d-printing #4: Enclosures & cablemanagement question 🤔

mglaman, 2021-05-19:
I bought your exact setup :) I have some questions on how you manage your cables for connections and power to everything. The pictures _don't quite_ show it.

The enclosure is really cool. But there is a cord port in the front left of the enclosure when the V2 has its power cable coming out of the back left. Honestly, the enclosure just feels backward for the V2 ( the flap on the right side would be _great_ on the other side for feeding in the filament.)

Do you run your cables out of that right side, or use their port?

geerlingguy, 2021-05-19:
@mglaman - Haha I thought of you when I started writing this repo's README!

I'm using that giant flap on the right side, I just peeled back a tiny bit of the corner and pass in one extension cord (with three outlets on it), and plug it all in there.

I have the LED light, the Ender 3 V2, and the Pi plugged into that cord (using separate sockets).

The weird thing is—and I'm not sure if it's because I'm using an AFCI breaker or what—every few weeks, when I start preheating the hot end or bed, the breaker will trip and my workshop outlets go dead... nothing else besides one 11W lamp is turned on, so I'm wondering if there's something about the way the relay works inside the Ender 3 V2 that trips the AFCI's fault and kills the circuit.

Anyways, way off topic, but you caught me after I spent 2 hours debugging what turned out to be a DNS issue... so my brain's a bit hazy :D

mglaman, 2021-05-19:
Ha! I was going to write a blog post review of the setup but I wanted to ping you about the case to see if I was nuts on the case setup. Interesting on the breaker tripping. That'll be a lovely experiment here (new house, basement finished, not easy to access any of the wiring 🙄 )

geerlingguy, 2024-07-10:
Going to close out this issue as I'm cleaning things up and will probably re-organize the whole repo a bit, since I've gone in less than a month from having one 3D printer to having five lol.

closed 2024-07-10 by geerlingguy