geerlingguy/3d-printing #6: Slicer?

wbreiler, 2022-01-31:
What slicer do you use? Noticed you use OctoLapse and I kept trying to get it to play nicely with Cura but no matter what I did it never seemed to like it

geerlingguy, 2022-01-31:
Ah, I should probably add some notes about my use of Cura. I did get it working with OctoLapse, after following these directions to add some Start G-Code: https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#if-you-are-using-cura-follow-these-steps

geerlingguy, 2024-07-10:
...and times change—I still use Cura for the Enders, but I've started using BambuStudio for the Bambu P1S and PrusaSlicer for the MK4. I may settle back in to just using Cura, but there are some benefits to the vendor-specific slicers!

wbreiler, 2024-07-10:
I’m not sure about the MK4, but the Enders should have a profile in OrcaSlicer, which is a fork of Bambu, which is then a fork of Prusa.

geerlingguy, 2024-07-10:
Heh, it gets confusing to even know what the heritage of these slicers is anymore!

wbreiler, 2024-07-10:
It all started in the dark ages, with Slic3r…

geerlingguy, 2025-01-16:
I'm also now using Orca for the Positron V3.2! So many slicers lol

wbreiler, 2025-02-14:
If you decide to try out [Input Shaper](https://help.prusa3d.com/article/input-shaper-mk4-s-mk3-9-s-mk3-5-s-xl-mini_451816), there's a profile in OrcaSlicer for the MK4(non-S).

geerlingguy, 2025-02-15:
After Bambu's firmware stunts, I've stopped using BambuStudio entirely, and just use OrcaSlicer directly with the P1S in Lan Mode. Seems to work great!