geerlingguy/3d-printing #9: Get Camera Module 3 working on Pi 5 with OctoPrint

geerlingguy, 2024-06-07:
I'm setting up a new Pi 5 with OctoPrint—well, hopefully—but I've run into a snag. I have a Camera Module 3 attached, but the latest nightly (6-7-2024) image build isn't working out of the box.

I started debugging over in this issue: https://github.com/guysoft/OctoPi/issues/823#issuecomment-2155646805

But rather than use that issue for rubber-duck-debugging, I'll use this issue to not pollute the OctoPi repo further, until I have more to report.

To debug, I just set up a brand new Pi 5 with a brand new Pi OS Bookworm 64-bit Lite install.

I ran apt update/upgrade, and installed the libcamera-apps suite: `sudo apt install libcamera-apps -y`. I rebooted, and verified I could see the camera:

```
pi@pistream:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

Then I tried using `go2rtc` to stream to my browser:

```
wget https://github.com/AlexxIT/go2rtc/releases/download/v1.9.3/go2rtc_linux_arm64
chmod +x go2rtc_linux_arm64

---

# Put the following inside go2rtc.yaml
streams:
  picam_h264: exec:rpicam-vid -t 0 --codec h264 -o -
  picam_mjpeg: exec:rpicam-vid -t 0 --codec mjpeg -o -

---

# Then run go2rtc:
./go2rtc_linux_arm64
```

In the browser, I opened http://10.0.2.204:1984 and opened one of the two streams. I end up with a window like:

[image]

or for mjpeg:

[image]

geerlingguy, 2024-06-07:
I found this thread: https://github.com/AlexxIT/go2rtc/issues/881 — it recommends using:

```
streams:
  picam_h264: exec:rpicam-vid -t 0 --inline --libav-format h264 -o -
  picam_mjpeg: exec:rpicam-vid -t 0 --codec mjpeg --libav-format mpegts -o -
```

But there was no difference. Still got the same result as above. I'm running Safari... could there be a weird issue there?

geerlingguy, 2024-06-07:
If I try opening the MJPEG stream (`http://10.0.2.204:1984/api/stream.mjpeg?src=picam_mjpeg`) directly from VLC, I get:

```
18:41:06.303 ERR [api.mjpeg] add consumer error="streams: exec/pipe: %!w(<nil>)\n[0:01:09.318561982] [940] \x1b[1;32m INFO \x1b[1;37mCamera \x1b[1;34mcamera_manager.cpp:284 \x1b[0mlibcamera v0.2.0+120-eb00c13d\n[0:01:09.328919040] [945] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mpisp.cpp:695 \x1b[0mlibpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)\n[0:01:09.434117733] [945] \x1b[1;32m INFO \x1b[1;37mRPI \x1b[1;34mpisp.cpp:1154 \x1b[0mRegistered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0\nPreview window unavailable\n[0:01:09.43900..."
```

geerlingguy, 2024-06-08:
Testing on a fresh Pi OS 12 install with [picamera2-WebUI-Lite](https://github.com/monkeymademe/picamera2-WebUI-Lite) instead:

```
# Install dependencies.
sudo apt install -y git python3-pip
sudo apt install -y python3-picamera2 --no-install-recommends

# Make Python not hate me.
sudo rm -rf /usr/lib/python3.11/EXTERNALLY-MANAGED

# Clone the project.
git clone https://github.com/monkeymademe/picamera2-WebUI-Lite.git
cd picamera2-WebUI-Lite
pip3 install Flask Pillow

# Run the app.
export PATH=$PATH:/home/pi/.local/bin
python app.py
```

And that works! Nice level of control (though the UI could be simplified and have like a 'beginner' mode with just a few focus, exposure, and WB controls).

[image]

glossyio, 2024-06-11:
> To debug, I just set up a brand new Pi 5 with a brand new Pi OS Bookworm 64-bit Lite install.

@geerlingguy - I had to use Install Raspberry Pi OS Bookworm (Full Install). Full Install required for all components to run properly. Lite is missing H.264 codec for libcamera.

glossyio, 2024-06-12:
I think you'd have to build [rpicam-apps source code](https://github.com/raspberrypi/rpicam-apps), and [rpi docs](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps).  I opted to do the Full Install vs. the build, so I couldn't help you beyond that.

AlexHuebi, 2024-10-24:
I got the camera working on Lite.
@glossyio is correct, Lite is missing some libraries. 
It can be fixed fairly easily though even without rebuilding the whole source code or installing RPiOS Full.

You'd have to use rpicam-apps instead of the preinstalled rpicam-apps-lite.

So with a fresh Raspberry Pi OS Lite install the following Commands are needed to get the camera running:

```
sudo apt update && apt upgrade -y 

sudo apt remove rpicam-apps-lite
sudo apt install rpicam-apps --no-install-recommends
``` 

and to get the Camera 3 Module Working with picamera2-WebUI-Lite: 
```
# install required packages
sudo apt install git python3-flask python3-pillow python3-picamera2

# download and start WebUI
git clone https://github.com/monkeymademe/picamera2-WebUI-Lite.git
cd picamera2-WebUI-Lite/
python3 app.py
```

jneilliii, 2025-01-13:
I would recommend looking at go2rtc as well instead of the picam2-webui. Here's a quick and dirty install script I did today for someone in the OctoPrint discord...adjust service files for non-`pi` usernames...

https://gist.github.com/jneilliii/d837afbf8a66c716547309f36a598bb0

jneilliii, 2025-03-23:
Wondered if anyone came across the [Arducam fork of mjpg-streamer](https://github.com/ArduCAM/mjpg-streamer) before? In theory it should be a drop-in replacement of the OctoPi webcamd service, just have to compile and replace the webcamd script maybe?

https://forums.raspberrypi.com/viewtopic.php?t=332154

OwlInSpace, 2025-04-18:
> Wondered if anyone came across the [Arducam fork of mjpg-streamer](https://github.com/ArduCAM/mjpg-streamer) before? In theory it should be a drop-in replacement of the OctoPi webcamd service, just [...]
I'm using a Prusa MK4S + Pi 5 + Cam 3. I tried all the "known" methods of getting this to work, and the TL;DR is that `go2rtc` worked the best and it would probably be better to switch to a Pi 4 w/ camera-streamer.

First, the ones that didn't work:

`camera-streamer`: I so wanted this to work as it had a plugin for webrtc support, but this obviously doesn't work as the Pi 5 doesn't have hardware acceleration

`ustreamer`: Seemed like really good software, and would probably similar to or better to `go2rtc` in terms of performance, but it doesn't work on 64-bit bookworm because of a bug in libcamerify, so you have to downgrade to Bullseye if you need it I think

The ones that did work and that I tried:

`Arudcam's mjpg-streamer`:  I ran this one in a docker container, it always uses an entire thread even when you put 1fps and 200x200 resolution for some reason and **is single-threaded** which means that it often stutters and lags as it can't keep up. The video feed is always on even when nobody was watching, wasting resources. If anyone wants my config for this one I'll upload it, I made some changes to a repo from 3 years ago to work with the latest packages

`Pi Camera 2 WebUI`: This one worked quite well, and the webui is very nice, but some stuttering was observed and this led to me look for alternatives. IIRC the video feed was always on too even when nobody was watching

`Go2RTC`: This streamed very smoothly and took the video feed off when it wasn't in use, which is a big plus.

jneilliii, 2025-04-19:
> The WebRTC stream in comparison is so perfect. I estimate around 800ms latency on the WebRTC stream and 3-4s on HLS. Uses ~200% of CPU

If you install my go2rtc plugin you can get webrtc support embedded in OctoPrint, similar to the camera-streamer control plugin.

Lasutriv, 2025-06-12:
> Testing on a fresh Pi OS 12 install with [picamera2-WebUI-Lite](https://github.com/monkeymademe/picamera2-WebUI-Lite) instead:

Thanks for this! Worked like a charm and was just what I needed for a simple quick solution