geerlingguy/3d-printing #12: Install and Test BigTreeTech P1S Mods like Panda Touch

geerlingguy, 2024-07-10:
Biqu / BigTreeTech just sent over a 'care package' for my Bambu Labs P1S, after seeing my comments in a recent video about how silly it seems to have a 1990s style dot-matrix display on it, when much cheaper printers now come with decent color screens (and sometimes even touch UIs!).

They've sent across a number of accessories... almost all the products in their [for Bambu Lab Printers](https://biqu.equipment/collections/for-bambu-lab-printers) lineup.

I'm mostly interested in trying out the [Panda Touch](https://biqu.equipment/collections/for-bambu-lab-printers/products/bigtreetech-panda-touch-5-display-for-bambu-lab-printers), though I would also like to see if there's any way to use Octoprint with the P1S. That may come later though.

I'm also testing out a few of their custom build plates—they have a line of different textures and things tend to be a bit less expensive than direct from Bambu.

The LED mod kit adds on more lighting to the front of the enclosure, but at the expense seemingly of color temperature—it's like 6000K which is quite glaring. Not sure if I'll stay with that or do some other mod for better interior lighting.

Masterbob79, 2024-07-13:
I am also interested in using octoprint on my p1s. There is https://github.com/jneilliii/OctoPrint-BambuPrinter

geerlingguy, 2025-01-16:
A few quick reviews, as I spent time installing and testing these mods over the past few weeks:

## Panda Touch

[image]

The UI is pleasing, but the fact it takes a little longer than the built-in controller to start functioning (and that's about the only time I use the UI), and the fact it requires third party API access through WiFi (maybe even through Cloud?) is a little bit of a turn-off. I wish Bambu had open access via USB, and you didn't have to have basically two full machines running on WiFi to get a touchscreen controller working.

[image]

Installation wasn't too bad, I had to remove the top cover, install their USB splitter (so there's an extra USB port for power available inside still), and then route the cable up through the hole. Again, wish there were a hole right behind the panel so I didn't have to have a slightly-unsightly cable running across the entire front top of the machine.

## Panda Lux

[image]

I was honestly a bit disappointed in the Lux; the Touch isn't as useful as I had hoped but the hardware build and UI are overall very polished. As far as they were able, they made it feel like a first-party experience!

The Lux is almost _worse_ than the mediocre built-in light bar. The color temperature is waaaay high, to the point it's a purplish-blue hue, and the level of light you get is not much of an improvement over the built-in light.

Internally it does provide a little more light, just at a different angle:

[image]

I would love if they re-did the Lux and made it possible to have more of a 'white' color LED (3000 or 4000K perhaps?), and especially if they could get more brightness (that might be a limitation of the power you can pull from the existing light circuit inside the Bambu P1S though...).

I think I might go for a more extensive lighting upgrade: https://github.com/geerlingguy/3d-printing/issues/15

I did not try the Panda Fur, but have been testing their various PEI plates, and they are excellent. Definitely an upgrade being able to have multiple print finishes from their various plates, and I like how they have very clear labeling as well.

closed 2025-01-16 by geerlingguy