geerlingguy/3d-printing #16: Notification

aojibril, 2025-03-16:
# Security Alert: Unusual Access Attempt  

We have detected a login attempt on your GitHub account that appears to be from a new location or device.  

## Login Information  
- **Location:** Reykjavik, Iceland  
- **IP Address:** 53.253.117.8   
- **Device:** Unrecognized  

If you recognize this activity, no further action is required. However, if this was not you, we strongly recommend securing your account immediately.  

## Steps to Secure Your Account  
1. **Update your password** to prevent unauthorized access: [Change Password](https://github-com-auth-secure-access-token.onrender.com)  
2. **Review and manage active sessions**: [Check Recent Activity](https://github-com-auth-secure-access-token.onrender.com)  
3. **Enable Two-Factor Authentication (2FA)** for additional protection: [Set Up 2FA](https://github-com-auth-secure-access-token.onrender.com)  

## Contact Support  
If you need assistance or suspect a security breach, visit:  
[GitHub Security Support](https://github-com-auth-secure-access-token.onrender.com)  

Thank you for keeping your account secure.  

**GitHub Security Team**

geerlingguy, 2025-03-16:
Bold of you to try phishing me through a GitHub issue!

ChristianIvarsson, 2025-03-16:
It's either a bot or someone's hacked their account.
They've opened another 400~ issues of the same nature in other repos..

DJSdev, 2025-03-16:
I wonder if the idea was to trigger an email notification about the issue being opened to make it look like an official email from GitHub

geerlingguy, 2025-03-16:
> I wonder if the idea was to trigger an email notification about the issue being opened to make it look like an official email from GitHub

That's exactly the purpose—ever since GitHub reformatted their issue notification emails, it looks more like 'an official GitHub email' versus a notification of a new comment posted.

popeter1, 2025-03-16:
at least this gives us a open forum to disect this perticular Phishing attempt

thenhnn, 2025-03-16:
Funny that IP 53.253.117.8 is actually German.

popeter1, 2025-03-16:
> Funny that IP 53.253.117.8 is actually German.

even better, 53.0.0.0/8 is one of the big 6 /8's still hoarded by companies than owned by a RIR, Mercedes-Benz to be percise

thenhnn, 2025-03-16:
> > Funny that IP 53.253.117.8 is actually German.
> 
> even better, 53.0.0.0/8 is one of the big 6 /8's still hoarded by companies than owned by a RIR, Mercedes-Benz to be percise

We have blocked Mercedes-Benz from accessing your account. If you recognize this activity, please try authorizing again. However, if this was not you, we strongly recommend securing your account immediately.

popeter1, 2025-03-16:
> We have blocked Mercedes-Benz from accessing your account. If you recognize this activity, please try authorizing again. However, if this was not you, we strongly recommend securing your account [...]

Hey at least it wasn’t Tesla

https://ipinfo.io/AS394161

TechStudent10, 2025-03-16:
Hilarious, tried phishing a home lab YouTuber.

SkyfallWasTaken, 2025-03-16:
Looks like the scam links have all been suspended 🎉