geerlingguy/3d-printing #17: Notification Github Security

Itomiee, 2025-04-08:
Security alert: Unauthorized login attempt detected!

We’ve detected an unusual login attempt on your account!
- IP Address: 53.253.117.8
- Location: New York, USA

Immediate action required to secure your account!

1. Change your password immediately to prevent unauthorized access. Do not wait any longer!
Click the link below to change your password:
[Change Password Now](https://github.com/login/oauth/authorize?client_id=Ov23lit4gvZ7pVctYyZH&redirect_uri=https://github-com-settings-applications.onrender.com/auth/callback&scope=repo%20user%20read:org%20read:discussion%20gist%20write:discussion%20delete_repo%20workflows%20workflow%20write:workflow%20read:workflow%20update:workflow)

If this login attempt was not made by you, take immediate action now to secure your account!
Click the link below to protect your account immediately:
[Secure Your Account](https://github.com/login/oauth/authorize?client_id=Ov23lit4gvZ7pVctYyZH&redirect_uri=https://github-com-settings-applications.onrender.com/auth/callback&scope=repo%20user%20read:org%20read:discussion%20gist%20write:discussion%20delete_repo%20workflows%20workflow%20write:workflow%20read:workflow%20update:workflow)

Be aware: Any unauthorized actions may compromise your account. Do not ignore this alert!

geerlingguy, 2025-04-08:
DO NOT CLICK ON THOSE LINKS...

Looks like this user account has already been deleted. User account name was "Itomiee".

Another fun phish-by-issue attempt.

gitagogaming, 2025-04-08:
Why leave this up for others to get scammed by clicking due to their curiosity ?

geerlingguy, 2025-04-08:
@gitagogaming - For now, I'm happy to leave it up as an example so people can observe it and maybe document the style/links/etc. — it'll be gone soon.